eclipse-4diac/4diac-forte #17: STRUCT_MUX crash

franz-hoepfinger-4diac, 2024-02-22:
STRUCT_MUX crash when there is a Struct which has a Namespace

cochicde, 2024-05-14:
@franz-hoepfinger-4diac can you check with the latest develoment branch if the problem is fixed for your case?

diplfranzhoepfinger, 2024-06-02:
@cochicde i can, but i have in the meantime removed the namespace, 
so i will have to add it again. Please hold on, i will do, but not immediately.

MartinMelikMerkumians, 2024-07-19:
Any news here? If there is no activity until next Friday, I will close this issue

diplfranzhoepfinger, 2024-07-19:
Ok. I can test since then.

franz-hoepfinger-4diac, 2024-07-23:
now it shows one warning in addition: 

[image]

franz-hoepfinger-4diac, 2024-07-23:
i get 2 Warnings now in IDE: 

```
Description	Resource	Path	Location	Type
Unused import 'utils::conversion_struct::types::bits::ST04B'	Uebung_051.SUB	/test/FBs/Uebungen	Uebung_051/compilerInfo/imports	4diac IDE Import Problem
Unused import 'utils::conversion_struct::types::bits::ST04B'	Uebung_052.SUB	/test/FBs/Uebungen	Uebung_052/compilerInfo/imports	4diac IDE Import Problem
```

azoitl, 2024-07-23:
> now it shows one warning in addition: [...]

This is not a warning this is an error!

franz-hoepfinger-4diac, 2024-07-23:
> > now it shows one warning in addition:
> This is not a warning this is an error!

@azoitl Thanks, i updated it. 
Obviously it is an Error. 
Not a Warning.

azoitl, 2024-07-23:
> > > now it shows one warning in addition:
> > > This is not a warning this is an error!
> 
> @azoitl Thanks, i updated it. Obviously it is an Error. Not a Warning.

I was to quick with my response. I wanted to write that `-` is not a valid character in an IEC 61131-3 identifier. The individual segments of an package have to be a IEC 61131-3 identifier.

franz-hoepfinger-4diac, 2024-07-23:
[image]

no more crashing, even if the Struct is of Type: `utils::conversion_struct::types::bits::ST04B`

franz-hoepfinger-4diac, 2024-07-23:
> I was to quick with my response. I wanted to write that `-` is not a valid character in an IEC 61131-3 identifier. The individual segments of an package have to be a IEC 61131-3 identifier.

yes, and that meant: 
i had to rename the Folder: 

https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/4diac_training1/tree/main/Ventilsteuerung/4diacIDE-workspace/test/Type%20Library/utils/conversion-struct

to 

https://github.com/Meisterschulen-am-Ostbahnhof-Munchen/4diac_training1/tree/main/Ventilsteuerung/4diacIDE-workspace/test/Type%20Library/utils/conversion_struct

and then the generated Package Name is right. 

other way would have been to edit the Package name manually while not overtaking what the IDE recommends.

PR #112 (2024-05-13): Remove call to log with nullptr and cleanup GenStructMux
closes #17 

- Avoid calling `DEVLOG_ERROR("%s", nullptr);`
- Cleanup function with modern cpp features
- Keep happy path to the left for a clearer view of the flow
closed 2024-07-23 by franz-hoepfinger-4diac