eclipse-4diac/4diac-forte #24: E_CYCLE crashes with assertation on windows.

murinsel, 2024-02-26:
The problem is a corrupt vector iterator in void CTimerHandler::processTimedFBList(). The problem definitly occurs if only one E_CYCLE is used. This results in the runtime assert:
[image]

The following code fixes this:

```
void CTimerHandler::processTimedFBList() {
  bool fired;
  do {
    fired = false;
    // We keep a list of sorted entries (oldest at the beginning) so we only have to check the first entry in the list:
    auto it = mTimedFBList.begin();
    if (it != mTimedFBList.end() && it->mTimeOut <= mForteTime) {
      STimedFBListEntry entry = *it; // buffer entry
      it = mTimedFBList.erase(it);
      triggerTimedFB(entry);  // This adds the entry again if not a one shot.
      fired = true; // Check the next entry.
    }
  } while (fired);
}
```

azoitl, 2024-02-26:
Thx a lot for finding this. Could you submit your change as PR, so that you can also get credit for it. I only find it strange that it didn't happen to me so far, because I'm regularly testing with only one E_CYCLE.

murinsel, 2024-02-26:
Could be that this is only a problem on Windows? I will try my first PR here.

azoitl, 2024-02-26:
It is maybe only observed on Windows. But it is definitely an issue because the triggerTimedFB is modifying the list while we are at the same time iterating over it. This not a good behavior. However I hoped to get rid of the iterative erase and do a full erase of all timed entries after the loop. But I fear this will also not be possible. I may have to talk with some C++ experts here.

murinsel, 2024-02-27:
I am getting always error 403 when trying to push. I have created a audentication token but that did not help.

azoitl, 2024-02-28:
In github you have to pull to your fork and then open a pull request. Based on the documentation of the Eclipse Platform I tried to explain it in our https://github.com/eclipse-4diac/4diac-forte/blob/release/CONTRIBUTING.md file. 

In the meantime I discussed with C++ experts. Unfortunately both of our solutions have performance drawback. If you agree I would push something later today.

murinsel, 2024-02-28:
Ok, fine.

azoitl, 2024-03-01:
> Ok, fine. I think using a vector here is problably not the best solution performace wise. Why was this changed from the self handled list implementation in the 2.x version of FORTE that did not have this problems?

Performance wise a vector is much better then the list. Especially as it is a sorted vector inserting and searching is much faster.

With PR #26 this issue should now be solved. @murinsel can you please confirm?

murinsel, 2024-03-03:
I have tried it an it works.

azoitl, 2024-03-03:
Great. Then I'll close this issue.

PR #26 (2024-03-01): Fix concurent modification of timerlist
Periodic entries in the timer list, wich are readded after triggering into the list resulted in a concurent modifcation. With this change the list is handled correctly and als copying of entries is reduced.

Fixes #24
closed 2024-03-03 by azoitl