eclipse-4diac/4diac-forte #59: FORTE compatibility to Windows 7

kumajaya, 2024-03-22:
The GetSystemTimePreciseAsFileTime function from commit [#4b5dd43290bf3ef97303c022e6acb2fb646e36d2](https://github.com/eclipse-4diac/4diac-forte/commit/4b5dd43290bf3ef97303c022e6acb2fb646e36d2) requires a minimum of Windows 8 which is the only function that prevents FORTE runs on Windows 7. I made a modification and replaced it with GetSystemTimeAsFileTime (configurable via CMake with WINDOWS_7_COMPAT definition) but not sure if this is the correct way to fix it regarding the expected level of precision.

kumajaya, 2024-03-23:
Proposed fix as a PR: https://github.com/eclipse-4diac/4diac-forte/pull/60

azoitl, 2024-03-29:
With PR #60 we have a very nice generic fix for that issue. Thx to all supporters.

kumajaya, 2024-03-30:
Thanks to all for the great guidance.

closed 2024-03-30 by kumajaya