eclipse-4diac/4diac-forte #83: FORTE with static MQTT on windows fails to link

ernstblechaPT, 2024-04-16:
After seeing #54 I have tried to get a fully static-linked FORTE on Windows. 

I have successfully built PAHO-MQTT with `PAHO_BUILD_SHARED=OFF` and `PAHO_BUILD_STATIC=ON` and changed the FORTE configuration in CMake to use `libpaho-mqtt3a-static.a`

I also had to make the following changes: https://github.com/ernstblechaPT/4diac-forte/commit/75dee041ae679aec26178a3a6ee61716cd37699a otherwise linking would fail with
```
C:/forte/mingw32/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld: C:/forte/paho-mqtt/build_win/src\libpaho-mqtt3a-static.a(Base64.c.obj):Base64.c:(.text+0x37): undefined reference to `__imp_CryptStringToBinaryA'
C:/forte/mingw32/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld: C:/forte/paho-mqtt/build_win/src\libpaho-mqtt3a-static.a(Base64.c.obj):Base64.c:(.text+0x76): undefined reference to `__imp_CryptBinaryToStringA'
collect2.exe: error: ld returned 1 exit status
```

Do you have any pointers where I went wrong?

kumajaya, 2024-04-16:
To be honest, the instructions weren't very clear. For gcc, you only need to add "-lws2_32 -lcrypt32 -lbcrypt" to CMAKE_CXX_STANDARD_LIBRARIES to link FORTE to modbus, paho, open62541 static library or "ws2_32.lib crypt32.lib bcrypt.lib" for MSVC compiler.

kumajaya, 2024-04-22:
@ernstblechaPT With a small changes, I just pushed your commit as part of PR https://github.com/eclipse-4diac/4diac-forte/pull/96

closed 2024-05-06 by ernstblechaPT