eclipse-4diac/4diac-forte #99: Is there any consideration to using human readable time in DEVLOG?

kumajaya, 2024-04-24:


azoitl, 2024-04-24:
So far not. But I like the idea.

kumajaya, 2024-04-25:
> So far not. But I like the idea.

Sir, something like https://github.com/eclipse-4diac/4diac-forte/pull/100 ?

ernstblechaPT, 2024-04-25:
My 2 cents:
 - dropping the resolution down to milliseconds might be problematic since errors are now logged on the error-output-stream and all other messages are on the standard-output, this might lead to confusion due to the streams being out of order
 - it would probably be the most consistent option if the time format was the DATETIME format specified in IEC 61131-3
 - alternativly https://en.wikipedia.org/wiki/ISO_8601#Combined_date_and_time_representations could be utilized as it is easily parseable, but this would require the target device to have access to up-to-date timezone information

azoitl, 2024-04-25:
Yesterday, when I first read the issue, I had as first idea to use NOW and the `toString()` Method to get the `Date_And_Time`'s IEC 61131-3 literal as representation. I only was not sure about the performance implications.

But before we continue here we need to get the Zephyr code out of this file. Otherwise the code is unreadable.

kumajaya, 2024-04-25:
For the meanwhile, I convert the PR as a draft.

kumajaya, 2024-04-25:
> Yesterday, when I first read the issue, I had as first idea to use NOW and the `toString()` Method to get the `Date_And_Time`'s IEC 61131-3 literal as representation. I only was not sure about the [...]

No issue for Windows, Linux, or embedded Linux but not sure about resource constrained device. For Zephyr, it does not use the standard chrono time functions.

kumajaya, 2024-04-26:
Add switch, optional logger time in IEC 61131-3 date time literal.

kumajaya, 2024-04-29:
PR https://github.com/eclipse-4diac/4diac-forte/pull/100 has been merged. Thank you all for the great guidance.

closed 2024-04-29 by kumajaya