eclipse-4diac/4diac-forte #103: Commit "DEVLOG: Option for human readable time" breaks build

kaartee, 2024-05-02:
At least on Zephyr targets, the mentioned commit (3d80e37550649e0c777d7c351c7ab72a7913e6e0) breaks the build, caused by two quality issues:

There is no proper signature in a header file for `std::string getRealtimeString()`, instead just an ad-hoc in  `src/arch/devlog.cpp`. This in turn relies on the class signature for `std::string` which is incorrectly assumed to be available indirectly via some other pre-existing includes.

Please fix the prototyping for `getRealtimeString()`, and properly include `string.h`.

kumajaya, 2024-05-02:
@azoitl Sir, do you have any suggestions for this issue?

azoitl, 2024-05-02:
> @azoitl Sir, do you have any suggestions for this issue?

Yes, first if all we need to separate the different platforms. There is a reason I recommended to solve the problem not with ifdefs. They do no scale. Therefore the first step is to split from generic code and platform specific. And here especially move the Zephyr code to a different Zephyr specific file. 

I also strongly disagree that ` std::string getRealtimeString()` should be declared in a header file. I don't want this to be api for something else then the log printer.

kaartee, 2024-05-03:
> I also strongly disagree that ` std::string getRealtimeString()` should be declared in a header file. I don't want this to be api for something else then the log printer.

I for one believe that in C++ prior to modules it is an anti-pattern to use non-internal linkage functions without declaration in a header file. Just because the C++ compiler checks name mangling, doesn't mean it's a good idea to share signatures via copy&paste between compilation units. The existence of a header file shouldn't be construed as an invitation to use it without giving it some thought and consulting the documentation?

Perhaps the changes required for the suggested split into platform specific code will yield new headers etc anyway, where `getRealtimeString()` finds a proper home for it's signature declaration.

kaartee, 2024-05-03:
FYI, see changes below. This (now on current Zephyr release 3.6) builds. Hm. I don't have time to test the results regarding actually seeing any log messages via iostreams, though. I'll report once I get there.
```
diff --git a/src/arch/devlog.cpp b/src/arch/devlog.cpp
index e2f3ec0b..94f1cb41 100644
--- a/src/arch/devlog.cpp
+++ b/src/arch/devlog.cpp
@@ -33,11 +33,7 @@
 #  include <cinttypes>
 # endif //< stdc11

-#ifndef __ZEPHYR__
 #include <iostream>
-#else
-#include <zephyr/logging/log.h>
-#endif // __ZEPHYR__

 #ifdef FORTE_STACKTRACE
 #include <iostream>
@@ -99,14 +95,10 @@ void logMessage(E_MsgLevel paLevel, const char *paMessage, ...) {
 }

 void printLogMessage(E_MsgLevel paLevel, const char *paMessage) {
-#ifndef __ZEPHYR__
   (paLevel == E_MsgLevel::Error ? std::cerr : std::cout) << scLogLevel[static_cast<int>(paLevel)] << ": " << getRealtimeString() << ": " << paMessage;
   #ifdef WIN32
   (paLevel == E_MsgLevel::Error ? std::cerr : std::cout) << std::flush;
   #endif
-#else
-  LOG_PRINTK("%s: T#%" PRIuFAST64 ": %s", scLogLevel[static_cast<int>(paLevel)], getNanoSecondsMonotonic(), paMessage);
-#endif // __ZEPHYR__
 }

 # endif  /* FORTE_EXTERNAL_LOG_HANDLER */
```

kaartee, 2024-05-03:
By the way, the PR underlying this issue failed to adapt the Zephyr preprocessor section, anyway. I don't think that was intentional, but...

kaartee, 2024-05-03:
I have tested both with and without setting `FORTE_LOGGER_READABLE_TIME` now, on my Zephyr target, Zephyr version 3.6, including the revert of the ZEPHYR special preprocessor handling above.

Old style (`-DFORTE_LOGGER_READABLE_TIME=OFF`):
```
INFO: T#560600000: FORTE is up and running
```
New style (`-DFORTE_LOGGER_READABLE_TIME=ON`):
```
INFO: DT#1970-01-01-00:00:00.551: FORTE is up and running
```

kaartee, 2024-05-08:
@azoitl To your knowledge, is anyone working on the more extensive changes you proposed? Otherwise, would you like me to issue an PR with only the removal of #ifdefs for Zephyr that I posted above?

azoitl, 2024-05-08:
> @azoitl To your knowledge, is anyone working on the more extensive changes you proposed? Otherwise, would you like me to issue an PR with only the removal of #ifdefs for Zephyr that I posted above?

@kaartee I wanted to but unfortunately my day job took over and I haven't found time yet. Would really appreciate any support on that.

azoitl, 2024-05-13:
> @azoitl One decision to make, in my mind, is whether C++ iostream shall remain a requirement for supported platforms. Regardless, could you advice on this, in `devlog.cpp`:
> 
> ```
> #include <iostream>
> 
> #ifdef FORTE_STACKTRACE
> #include <iostream>
> #ifdef FORTE_STACKTRACE_BOOST
> #include <boost/stacktrace.hpp>
> #endif // FORTE_STACKTRACE_BOOST
> 
> #ifdef FORTE_STACKTRACE_CXX23
> #include <stacktrace>
> #endif // FORTE_STACKTRACE_CXX23
> #endif // FORTE_STACKTRACE
> ```
> 
> I would remove the second, conditional, `#include <iostream>`?

Thx @kaartee  for your PR and the work. For now I would go with `#include <iostream>` in the default implementation. We allready have the option that a specific platform can provide an own logging infrastructure.

closed 2024-05-25 by azoitl