eclipse-4diac/4diac-forte #154: Using OPC UA results in 100% CPU core usage

gafert, 2024-06-11:
When using OPC UA in FORTE a single core is using 100%. 

I noticed this when using PowerLink and OPC UA. In combination PowerLink throws `0x8243` which corresponds to `EPL_E_DLL_LOSS_PRES_TH` as soon as a OPC UA node is part of the system. I guess it can't keep up with the cycle time which results in a restart of the PowerLink devices.

Reference to the PowerLink to the PowerLink `AppCbEvents`:

[image]

Investigating the PowerLink errors I noticed that the error is only thrown when an OPC UA node is part of the system. At the same time the CPU usage is stuck at a 100% when using OPC UA. Removing all PowerLink nodes in the system and recompiling with only OPC UA features also shows the same CPU usage. With only PowerLink the CPU usage was always lower than 10% and no errors are thown. Recompiling FORTE and open62541 did not change anything.

High CPU usage when ONLY using OPC UA (no PowerLink here):

[image]

# Reproduction

1. Build Forte with OPC UA support following https://eclipse.dev/4diac/documentation/html/communication/opc_ua_version1.html

```
cmake -DCMAKE_BUILD_TYPE=Release -DFORTE_ARCHITECTURE=Posix -DFORTE_MODULE_CONVERT=ON -DFORTE_COM_ETH=ON -DFORTE_MODULE_IEC61131=ON -DFORTE_COM_OPC_UA=ON -DFORTE_COM_OPC_UA_INCLUDE_DIR=../../open62541/build -DFORTE_COM_OPC_UA_LIB_DIR=../../open62541/build/bin -DFORTE_COM_OPC_UA_LIB=libopen62541.so ..
```

3. Create a System using OPC UA

[image]

4. Create `.fboot` and start FORTE with it

``` bash
admin@X1TBEES03: ~/forte/build/src$ ./forte 
INFO: T#1212520515334851: FORTE is up and running
INFO: T#1212520515468667: Using provided bootfile location set in CMake: forte.fboot
INFO: T#1212520515565327: Boot file forte.fboot opened
INFO: T#1212520515970966: Bootfile correctly loaded
INFO: T#1212520516041645: Closing bootfile
INFO: T#1212520516393086: [OPC UA LOCAL]: Starting OPC UA Server: opc.tcp://localhost:4840
[2024-06-11 14:47:04.138 (UTC+0200)] warn/server	Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
INFO: T#1212520530551869: [OPC UA LOGGER] info/network	TCP network layer listening on opc.tcp://BR-forte_4840:4840/
```

5. Observer high CPU usage

Running with Linux on an Intel Atom E3845 1,91 GHz Quad Core B&R PC with 8 GByte SDRAM.

cochicde, 2024-06-12:
I tried it now with the latest develop and get just around 4% usage. which branch in forte are you using and which open62541 version?

gafert, 2024-06-16:
Thanks for giving me hope. I had an older version (1.13.0) and tried it again with the release branch and it worked. The issue is therefore resolved.

While doing that, I noticed that powerlink does not compile in the release branch. The problem ([missing include dir](https://github.com/eclipse-4diac/4diac-forte/blame/d42a30cbead291c459c163d92212880774643f42/src/com/powerlink/CMakeLists.txt#L64)) is already fixed in develop. I then also saw that develop has a lot of work done in the opc ua module. 

So what branch should I end up using? Is develop stable enough? Or is [4diac-fbe](https://github.com/eclipse-4diac/4diac-fbe) the way to go (I found that by accident, I don't think it's documented on the website)? Also regarding the documentation of what external package versions each branch should use. For example, the docs say to use version 1.0 of open62541 but in fbe the version used is 1.3.9.

closed 2024-06-16 by gafert

cochicde, 2024-06-16:
> Thanks for giving me hope. I had an older version (1.13.0) and tried it again with the release branch and it worked. The issue is therefore resolved.
>
> While doing that, I noticed that powerlink does [...]

great to see that it works now! :). By definition the `develop` branch is the one chaging the most, but I think is quite stable still. The automatic checks try to guarantee that. The problem is that these automatic tests are not testing all modules so these modules might become unstable.  For OPC UA as you said there were some work done, but the module is properly tested. I never compiled or worked with powerlink so I don't know how hard it is to add it to the automatic tests, but having it there would make more stable.  

https://github.com/eclipse-4diac/4diac-fbe seems like a good place to start. I haven't use it personally myself, but it seems it has all the things needed. 

@azoitl might help you with more information about the status of the branches

azoitl, 2024-06-16:
1.13 is quite an old version. Lots of improvement have been done since then. Currently we are working on getting everything stable enough (mostly 4diac FORTE API stability) for an 3.0 release. As much stuff is going on we have currently little time to update our documentation. Which is unfortunately our weakest spot.

If you want to use OPC UA I strongly recommend to use 4diac FORTE freeze branch. This holds our intermediate test releases. The freeze branch should always compile and be stable enough for usage. Although we also have users working with the development branch.

However as a lot has changed in 4diac FORTE you need to use an according 4diac IDE for generating the C++ code for your FBs. We provide prebuilt, 4diac IDE freeze versions here: https://download.eclipse.org/4diac/updates/milestone/
When using this 4diac IDE version you will be automatically informed about updates of 4diac IDE and you can just upgrade. Your existing 4diac IDE project can be opened with that version. If that is not the case please let us know that we can fix that. 

If you have feedback for our documentation you can easily edit it here: https://github.com/eclipse-4diac/4diac-documentation
We just recently migrated to AsciiDoc, which allows for a very easy edit. You can directly do that on github and turn your improvements into PRs.

gafert, 2024-06-18:
@azoitl Thanks for the feedback and the link to the IDE milestones. 

While trying the freeze branch it took me a while to figure the `UA_*` cmake arguments required which I later found in https://github.com/eclipse-4diac/4diac-fortebuildcontainer/blob/bbeb99c0ef41130cdf48923a9540158196f55416/Dockerfile#L33. Is there a place to put documentation of the current develop branch? That would also help the 3.0 release.

azoitl, 2024-06-18:
@gafert yes please have a look at the documentation in https://github.com/eclipse-4diac/4diac-documentation. I guess the OPC UA documenation would be the right place for such information: https://github.com/eclipse-4diac/4diac-documentation/blob/main/src/communication/opcUA.adoc

But if you have a better place feel free to tell and PR us.