eclipse-4diac/4diac-forte #169: Powerlink module is not compileable

gafert, 2024-06-20:
The Powerlink module is not compileable in the latest development branch.

The FBs don't match the required format anymore and need to be regenerated and adapted. Furthermore, commit https://github.com/eclipse-4diac/4diac-forte/commit/3cd52d7c9d66c7923f6f1f2973bbec64b91f3d2a renamed all variables which resulted in unresolved variables in `EplWrapper.cpp` as the names in the structs do not match the struct definition included from openPOWERLINK.

I am currently implementing a fix and can create a merge request if wanted.

# Reproduction

Download and build powerlink lib (.lib never used as we don't get to the linking stage, but here for completeness)
```
git clone --depth 1 --branch v1.08.5 https://github.com/OpenAutomationTechnologies/openPOWERLINK_V2 powerlink
cd ./powerlink/ && mkdir build && cd $_
cmake .. && cmake --build .
mv ./Examples/X86/Generic/powerlink_user_lib/libpowerlink.a ../libpowerlink.a # lib in root of powerlink expected
cd ../.. # Back to main folder
```

Download tinyxml
```
git clone https://git.code.sf.net/p/tinyxml/git tinyxml
```

Download and build forte
```
git clone --depth 1 --branch develop https://github.com/eclipse-4diac/4diac-forte forte
cd ./forte/ && mkdir build && cd $_
cmake -DFORTE_ARCHITECTURE=Posix -DFORTE_MODULE_POWERLINK=ON -DFORTE_MODULE_POWERLINK_LIB_DIR=../../powerlink -DFORTE_MODULE_POWERLINK_TINYXML_DIR=../../tinyxml ..
cmake --build .
```

Build throws errors as FBs are not compatible and later unresolved variables.

azoitl, 2024-06-20:
It would be great if you could fix the issues and provide PRs. If possible and if it makes sense feel free to split the work in several smaller ones.

azoitl, 2024-06-24:
What I forgot in my last message: As you are anyhow working on it you may also want to consider to migrate it to the new modular IO infrastructure. This would make it much easier for you to use and decouple your application code from the network setup.

diplfranzhoepfinger, 2024-07-22:
> What I forgot in my last message: As you are anyhow working on it you may also want to consider to migrate it to the new modular IO infrastructure. This would make it much easier for you to use and decouple your application code from the network setup.

I support it very much. Having Powerlink in new IO would be cool.

gafert, 2024-09-13:
Fixed in #171

closed 2024-09-13 by gafert

azoitl, 2024-09-13:
@gafert thx for spending the effort in upgrading and improving the powerlink implemenation