eclipse-4diac/4diac-forte #220: Problem when exporting CFBs with IO blocks

MandKastner, 2024-08-28:
there is a Problem with Composite FBs. 

`    forte::core::CInternalFB<FORTE_IX> fb_IX;`

it then does not find the 

`#include "IX_fbt.h"`

and if you specifiy the Path in CMakeList.txt, 

then it does not find the 

`#include "processinterface.h"`

and if you specifiy this Path  as well in CMakeList.txt, 

then it does not link. 

maybe it is a bit special to have a IX in a Composite FB, but maybe this points to a special Error, which should be solved anyhow. 

thanks, 
Franz

_Originally posted by @diplfranzhoepfinger in https://github.com/eclipse-4diac/4diac-ide/issues/360#issuecomment-2312753296_

azoitl, 2024-08-28:
@mx990 just transferred it. 

Thx for the fix. This a very nice workaround for now. In the long run (i.e., read when I find time) I want the `processinterface.h` to die.  Most probably this will only happen for modular IO but especially for modular IO the current solution is unnecessary complex and inefficient.

franz-hoepfinger-4diac, 2024-09-25:
this is solved to my Knowledge, right ?

azoitl, 2024-09-25:
To my understanding yes.