eclipse-4diac/4diac-forte #227: modbus-ng

diplfranzhoepfinger, 2024-09-13:
@mx990 

i wanted to have a look at Modbus-NG. 

[https://git.eclipse.org/r/c/4diac/org.eclipse.4diac.forte/+/139743](https://git.eclipse.org/r/c/4diac/org.eclipse.4diac.forte/+/139743)

can you please drop me a E-Mail, or help me here,  i am searching the FBs for Bus, Slave etc..

Thanks many times.

mx990, 2024-09-13:
Thank you for your interest in the modbus-ng I/O interface. I was unfortunately unable to locate the original FB type files from back then, but I quickly re-created them for you based on the generated code: https://github.com/eclipse-4diac/4diac-ide/pull/477

The Slave FB types are specific to the respective device, so I did not include them. The slave devices need to be SIFBs with the following interface:
```xml
		<EventInputs>
			<Event Name="MAP" Type="Event" Comment="Service Initialization">
				<With Var="QI"/>
<!--				<With Var="<IO Name>"/> -->
				<With Var="Unit"/>
				<With Var="UpdateInterval"/>
			</Event>
		</EventInputs>
		<EventOutputs>
			<Event Name="MAPO" Type="Event" Comment="Initialization Confirm">
				<With Var="QO"/>
				<With Var="STATUS"/>
			</Event>
			<Event Name="IND" Type="Event" Comment="Indication from Resource">
				<With Var="QO"/>
				<With Var="STATUS"/>
			</Event>
		</EventOutputs>
		<InputVars>
			<VarDeclaration Name="QI" Type="BOOL" Comment="Event Input Qualifier"/>
<!--			<VarDeclaration Name="<IO NAME>" Type="WSTRING" Comment="I/O Name"/> -->
			<VarDeclaration Name="Unit" Type="USINT" Comment="Modbus Unit Identifier"/>
			<VarDeclaration Name="UpdateInterval" Type="UINT" Comment="Update Interval"/>
		</InputVars>
		<OutputVars>
			<VarDeclaration Name="QO" Type="BOOL" Comment="Event Output Qualifier"/>
			<VarDeclaration Name="STATUS" Type="WSTRING" Comment="Service Status"/>
		</OutputVars>
```

The exported code then needs to be adapted to inherit from `ModbusSlave` and implement the required functionality, see also [SlaveFestoCMMP_AS.h](https://git.eclipse.org/r/c/4diac/org.eclipse.4diac.forte/+/139744/3/src/modules/modbus-ng/types/SlaveFestoCMMP_AS.h) and [SlaveFestoCMMP_AS.cpp](https://git.eclipse.org/r/c/4diac/org.eclipse.4diac.forte/+/139744/3/src/modules/modbus-ng/types/SlaveFestoCMMP_AS.cpp).

MartinMelikMerkumians, 2026-05-19:
Closed due to extended inactivity. If this is still relevant please comment and I will reopen the issue