eclipse-4diac/4diac-forte #242: Question about libforte-c-static.a vs. libforte-static.a

franz-hoepfinger-4diac, 2024-09-25:
Question about libforte-c-static.a vs. libforte-static.a

if i use: 
    -DFORTE_BUILD_STATIC_LIBRARY=ON \
    -DFORTE_C_INTERFACE=ON \

i get 2 Libs: 
libforte-c-static.a
libforte-static.a

is this intended ?

franz-hoepfinger-4diac, 2024-09-25:
it does also really consume CPU Time for Linking 2 libs

```
[100%] Linking CXX static library libforte-static.a
[100%] Built target forte-static
[100%] Linking CXX static library libforte-c-static.a
[100%] Built target forte-c-static

```

cochicde, 2024-09-25:
It was intended, yes. But just because thought that would be the usual case. Nothing is set in stone. 

So basically, as it is now, when you set `DFORTE_C_INTERFACE` to `ON` you get the `C` Libraries as extra. The other option would be that if you set it to `ON` then you build the C libraries instead of the regular `forte-static`.

I don't have any preference. The only con I see of the either/other option would be that you can't have both at the same time unless you configure twice. 

We could have a third variable that could decide on that, but I'm afraid it will get too complicated.

I think is just a matter of use case. If you think having both when `DFORTE_C_INTERFACE` is `ON` is a rare case, we could just change it.

@azoitl what do you think?

azoitl, 2024-09-26:
From my experience you would either have the c library or the other so some exclusive or switch could make sense. Especially if building gets much faster by that.

franz-hoepfinger-4diac, 2024-09-29:
definitely.

cochicde, 2024-09-30:
Allright! I can take a look at this, but only in a couple of weeks.

cochicde, 2024-10-28:
closing after #264 was merged

closed 2024-10-28 by cochicde