eclipse-4diac/4diac-forte #258: Clang/MacOS compiler error in opc-ua com layer

jwalt, 2024-10-07:
Clang on MacOS (via 4diac-fbe develop branch) fails in the OPC-UA layer:

```
In file included from /home/jwalter/lcl/4diac-fbe/build/test-macos/forte/build/com/opc_ua/opcua_types.cpp:12:
/home/jwalter/lcl/4diac-fbe/forte/src/com/opc_ua/opcua_helper.h:86:40: error: implicit instantiation of undefined template 'std::vector<const UA_Variant *>'
   86 |         std::vector<const UA_Variant*> mData;
      |                                        ^
/home/jwalter/lcl/4diac-fbe/toolchains/clang-toolchain/SDK/MacOSX11.1.sdk/usr/include/c++/v1/iosfwd:216:28: note: template is declared here
  216 | class _LIBCPP_TEMPLATE_VIS vector;
      |                            ^
In file included from /home/jwalter/lcl/4diac-fbe/build/test-macos/forte/build/com/opc_ua/opcua_types.cpp:12:
/home/jwalter/lcl/4diac-fbe/forte/src/com/opc_ua/opcua_helper.h:100:34: error: implicit instantiation of undefined template 'std::vector<UA_Variant *>'
  100 |         std::vector<UA_Variant*> mData;
      |                                  ^
/home/jwalter/lcl/4diac-fbe/toolchains/clang-toolchain/SDK/MacOSX11.1.sdk/usr/include/c++/v1/iosfwd:216:28: note: template is declared here
  216 | class _LIBCPP_TEMPLATE_VIS vector;
```

azoitl, 2024-10-07:
@cochicde could youplease have a look on that.

jwalt, 2024-10-07:
Just as an additional data point: It works fine when using clang for a Linux target. Must be something MacOS specific.

cochicde, 2024-10-07:
I cannot reproduce it locally, but could it be that we are missing a

#include <vector>

In that file?

jwalt, 2024-10-07:
You should be able to reproduce with:

```
git clone --recursive -b develop https://github.com/jwalt/4diac-fbe
# download https://github.com/joseluisq/macosx-sdks/releases/tag/11.1 and put it into toolchains subfolder
cd 4diac-fbe
./scripts/compile.sh test/test-macos
```

cochicde, 2024-10-07:
Thanks. I was able to reproduce and test the fix. See #259

cochicde, 2024-10-08:
closing as #259 was merged and fixes the issue

PR #259 (2024-10-08): add missing vector include
add missing include causing issues in macos/clang

Fixes #258
closed 2024-10-08 by cochicde