eclipse-4diac/4diac-forte #261: Impossible to compile composite FB that refers to OUT_ANY_CONSOLE_fbt.h

melekhove, 2024-10-14:
4diac-forte version: freeze branch.
FORTE_MODULE_UTILS:BOOL=ON

Compilation of the composite FB that contains any FB from modules/utils fails:
```
In file included from ...ext-modules/EXAMPLE_TEST/CompositeTest_fbt.cpp:12:
...ext-modules/EXAMPLE_TEST/CompositeTest_fbt.h:26:10: fatal error: OUT_ANY_CONSOLE_fbt.h: No such file or directory
   26 | #include "OUT_ANY_CONSOLE_fbt.h"
```
When the following line is added to the src/modules/utils/CMakeLists.txt compilation succeeds.
`forte_add_include_directories(${CMAKE_CURRENT_SOURCE_DIR})
`

mx990, 2024-10-14:
@melekhove Would you be willing to create a PR to fix the problem?

melekhove, 2024-10-15:
> @melekhove Would you be willing to create a PR to fix the problem?
Ok, please take a look at #262

azoitl, 2024-10-15:
Welcome to Eclipse 4diac. Thanks for the report and the fix!

melekhove, 2024-10-16:
> Welcome to Eclipse 4diac. Thanks for the report and the fix!
Thank you. Should I close the issue or someone else will ?

azoitl, 2024-10-16:
> Thank you. Should I close the issue or someone else will ?

We don't have yet established an issue closing policy. But as you opened it and fixed it I would like you to have to honor of closing it.

closed 2024-10-16 by melekhove