eclipse-4diac/4diac-forte #299: Can several FBs execute synchronously？

demon181, 2025-01-08:
Hi，all
I create a FB named PrintData, it printf 10 times-data in a loop with 5s. 
Then I drag 2 PrintData-FB in IDE,they all connect to E_RESTART COLD-pin parallel.
I wish all 2 FBs can print data synchronously, but actually, only 1st FB done the 2nd FB begin to execute then.
So can several FBs execute synchronously? if so, what's wrong do I meet, I make function executeEvent() to wait until Print-thread complete, does this not correct?  My Forte is 2.0.1.
thanks anyway.

azoitl, 2025-01-08:
IEC 61499 and 4diac FORTE assumes that FBs are algorithmic, short and mostly non blocking. The current implementation of 4diac FORTE uses per default one thread per resource and is delivering input events in a FIFO order. On event input delivery the FB is allowed to process until it is finished, when an FB sends output events they are put in the FIFO queue of the thread calling it. When the  FB has finished the next event is delivered and so on. There are ways to use more threads in a resource but I'm not sure if that is where you should go to. 

I don't know exactly your use case but IEC 61499 has dedicated FBs to provide you with timed events (i.e., E_DELAY, E_RDELAY, E_CYCLE, E_TIMEOUT). This can be used to do the tasks you need in a none blocking IEC 61499 event based way.

PS: I strongly encourage you to upgrade to the latest development version of 4diac FORTE it is much faster and uses less memory.

demon181, 2025-01-09:
Hello, actually I am a MotionController developer，I try to implement PLCOPEN FBs in 4diac now.
As learned from 61499 now, I wish an Event trig then X/Y axis can move, when they finished, they output done-Event to  trig another FB.
[image]
[image]
E_CYCLE periodic event, I think is not suitable for motion-control because of time-accuracy,maybe?
 Dose anyone else doing something already, I can not find more info in Web.

demon181, 2025-01-09:
hello azoitl,
as your suggestion,I add a time-event input in MoveAbs and E_CYCLE line to it to offer time,is this right?
Generally, E_CYCLE only provide period-time and REQ-event control the axis move or not.

azoitl, 2025-01-10:
Ah motion control. This is a different story. I think here it makes sense to investigate different approaches. Normally you have some kind of motion control core that accepts commands executes them and sends feedback back to the callers. We worked on such topics some time ago: https://link.springer.com/article/10.1007/s00502-006-0341

The question you have to consider is how you implement the motion control core. One option is to do it also in IEC 61499 (e.g., a resource with all the control loops and cycles) and have PUBL/SUBL blocks to interact with the PLCopen blocks.  

The other option is that you implement it in C++ and implement the PLCopen blocks as SIFBs. Here the C++ code should also run in its own seperate threads. The SIFB blocks would do some messaging to the C++ motion core and the core would send back external events so that the PLCopen SIFBs are not blocking. 

I hope this helps. I also would like to point you to our upcoming winterschool this could also be a place to discuss this: https://eclipse.dev/4diac/events/4diacwinterschool

Finally from your screenshots I noticed that you are using an older version of 4diac IDE. On our [Download PAge](https://eclipse.dev/4diac/download/) we are now finally offering preview versions of the upcoming 3.0 release. As we fixed and improved so many things (more then 5000 commits) I can only strongly recommend to upgrade.