eclipse-4diac/4diac-forte #350: Differences in Header of IO files

MoritzO02, 2025-03-09:
Some IO files have differences in People and Organisations

MoritzO02, 2025-03-09:
### revpi
Left is the old Header and on the right is the Header from IDE export. In the IDE only José Cabral exists as a persion with fortiss GmbH as Organisation

[image]

MoritzO02, 2025-03-09:
### embrick
Left is the old Header and on the right is the Header from IDE export. No person or organisation is given in the original Header

[image]

MoritzO02, 2025-03-09:
### embrick
Left is the old Header and on the right is the Header from IDE export. Difference in organisation for Johannes Messmer and José Cabral does not exist in IDE and has not organisation Info

[image]

MoritzO02, 2025-03-09:
### ID and QD
Left is the old Header and on the right is the Header from IDE export. Both have the same Header. Only Monika Wegner exists in IDE, the other two have no organisation Info 

[image]

azoitl, 2025-03-09:
> ### revpi
> 
> Left is the old Header and on the right is the Header from IDE export. In the IDE only José Cabral exists as a persion with fortiss GmbH as Organisation. The other People I added with manually and changed the Version numbers to reflect the old Header
> 
> [image]

azoitl, 2025-03-09:
Please don't do that for these cases. These are SIFBs where the interface has not changed in a long time but different people have been working on the C++ code for either changes in 4diac FORTE as a whole or just the IO subsystem. As I wrote in one of my other comments to you @MoritzO02, in these cases copyright headers differ and will continue to differ. It is wrong to harmonize them.

MoritzO02, 2025-03-10:
> Please don't do that for these cases. These are SIFBs where the interface has not changed in a long time but different people have been working on the C++ code for either changes in 4diac FORTE as a [...]

So, in all of these cases, keep the current forte Header and ignore the info from IDE?

azoitl, 2025-03-10:
> So, in all of these cases, keep the current forte Header and ignore the info from IDE?

Exactly!

MoritzO02, 2025-03-10:
In cases where all the Info is the same, should the original Header also be kept? In this instance the generated header has all the same info and adds a date to the contributor.

[image]

MoritzO02, 2025-03-11:
Should I also keep the original Header for QW/IW?

[image]

azoitl, 2025-03-11:
From the git history I don't see what I changed. Given all the small changes that happened in the last year I think at least the year in both (fbt, and c++) files should be updated to have 2015, 2025. But other than that I would keep it.

MoritzO02, 2025-03-11:
Okay, I'll add ,2025 to the cpp and h files

MoritzO02, 2025-03-11:
Where can I see the that there were changes in 2025? I can only see changes in 2019/2021 to the IW.cpp or QW.cpp in Github

azoitl, 2025-03-11:
> Where can I see the that there were changes in 2025? I can only see new changes in 2019/2021 to the IW.cpp or QW.cpp in Github

in the git history I see that the last minor changes where 4 months ago. So 2024 would be correct. But with your changes 2025 would be correct.

MoritzO02, 2025-03-13:
Okay, thanks. I think that's all for the io files and we can close this issue

azoitl, 2025-03-13:
For these small changes I normally do not add me to the name list. Because in the end you "only" changed the event type of the INIT/INIT pins.