eclipse-4diac/4diac-forte #362: Crash on replace:

franz-hoepfinger-4diac, 2025-03-15:
Crash on replace:

azoitl, 2025-03-15:
With the information given this is a bit a guessing game. Do you have some possibility to get a bit of debug information (e.g., line numbers) into your back trace. 

Could it be that this QX was never initialized, e.g., QI is not true? But with a second look on the code this is not possible.

As we have not changed anything here do you have a new compiler?

diplfranzhoepfinger, 2025-03-15:
My guess is I need to delete the handles. 

A delete for the handle is nowhere called. Also the handle is not deregistered. 

This I guess is the reason.

diplfranzhoepfinger, 2025-03-15:
Yes. Do you know how to get the Line Numbers? I guess I must set build_type to debug right?

azoitl, 2025-03-15:
> Yes. Do you know how to get the Line Numbers? I guess I must set build_type to debug right?

I think so.

diplfranzhoepfinger, 2025-03-15:
> My guess is I need to delete the handles.
> 
> A delete for the handle is nowhere called. Also the handle is not deregistered.
> 
> This I guess is the reason.

This did not help

franz-hoepfinger-4diac, 2025-03-15:
wuuuuuu ... that takes long with `CMAKE_BUILD_TYPE=DEBUG` and bunch of more flash Memory, Also Linking is veryyy long. 
now hope we will see Line Numbers !

franz-hoepfinger-4diac, 2025-03-15:
```
INFO: T#26000000000: Connection established by client
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060430  A0      : 0x822bdeba  A1      : 0x3fcdfd20  
A2      : 0x3c4b1a40  A3      : 0x00000000  A4      : 0x3c4ea7d4  A5      : 0x3c4e9764  
A6      : 0x3c4ea7c4  A7      : 0x3fcdfd20  A8      : 0x823e33f1  A9      : 0x3c4b1a40  
A10     : 0x3c4b1a40  A11     : 0x00000000  A12     : 0x3fca99bc  A13     : 0x3c4e9764  
A14     : 0x00000025  A15     : 0x3fcdfcf0  SAR     : 0x0000001a  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  
--- 0x40056f5c: __ssputs_r in ROM
0x40056f72: __ssputs_r in ROM

Backtrace: 0xfffffffd:0x3fcdfd20 0x422bdeb7:0x3fcdfd50 0x422be00b:0x3fcdfdb0 0x420d0f4e:0x3fcdfdd0 0x422bd976:0x3fcdfe20 0x422bd71d:0x3fcdfe50 0x420d42ad:0x3fcdfe80 0x420d42d6:0x3fcdfeb0 0x420db067:0x3fcdfee0 0x420db0c1:0x3fcdff20 0x420db067:0x3fcdff50 0x420db0c1:0x3fcdff90 0x420db067:0x3fcdffc0 0x420db0c1:0x3fce0000 0x420db067:0x3fce0030 0x420d9a61:0x3fce0070 0x420dd26d:0x3fce00a0 0x422adb13:0x3fce00d0 0x422adb2a:0x3fce0100 0x423ba8e3:0x3fce0130 0x420db3e1:0x3fce0160 0x420dd32b:0x3fce01b0 0x420d91b5:0x3fce0210 0x422add22:0x3fce0250 0x422ace75:0x3fce0290 0x422acddd:0x3fce02d0 0x420d98ed:0x3fce0350 0x420dfde7:0x3fce0380 0x420dfd47:0x3fce03c0 0x4201f33b:0x3fce03f0 0x4201f195:0x3fce0430
--- 0x422bdeb7: forte::core::io::ProcessInterface::dropHandle() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/io/processinterface.cpp:217 (discriminator 1)
0x422be00b: non-virtual thunk to forte::core::io::ProcessInterface::dropHandle() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/io/processinterface.h:81
0x420d0f4e: forte::core::io::IOMapper::deregisterObserver(forte::core::io::IOObserver*) at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/io/mapper/io_mapper.cpp:122
0x422bd976: forte::core::io::ProcessInterface::deinitialise() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/io/processinterface.cpp:88 (discriminator 1)
0x422bd71d: forte::core::io::ProcessInterface::~ProcessInterface() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/io/processinterface.cpp:39
0x420d42ad: FORTE_IX::~FORTE_IX() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/stdfblib/io/IX_fbt.h:47 (discriminator 14)
0x420d42d6: FORTE_IX::~FORTE_IX() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/stdfblib/io/IX_fbt.h:47
0x420db067: forte::core::CFBContainer::~CFBContainer() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/fbcontainer.cpp:58 (discriminator 1)
0x420db0c1: forte::core::CFBContainer::~CFBContainer() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/fbcontainer.cpp:61
0x420db067: forte::core::CFBContainer::~CFBContainer() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/fbcontainer.cpp:58 (discriminator 1)
0x420db0c1: forte::core::CFBContainer::~CFBContainer() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/fbcontainer.cpp:61
0x420db067: forte::core::CFBContainer::~CFBContainer() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/fbcontainer.cpp:58 (discriminator 1)
0x420db0c1: forte::core::CFBContainer::~CFBContainer() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/fbcontainer.cpp:61
0x420db067: forte::core::CFBContainer::~CFBContainer() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/fbcontainer.cpp:58 (discriminator 1)
0x420d9a61: CFunctionBlock::~CFunctionBlock() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/funcbloc.cpp:61 (discriminator 1)
0x420dd26d: CResource::~CResource() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/resource.cpp:83 (discriminator 2)
0x422adb13: EMB_RES::~EMB_RES() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/stdfblib/ita/EMB_RES.cpp:36 (discriminator 1)
0x422adb2a: EMB_RES::~EMB_RES() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/stdfblib/ita/EMB_RES.cpp:36
0x423ba8e3: CTypeLib::deleteFB(CFunctionBlock*) at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/typelib.cpp:175 (discriminator 1)
0x420db3e1: forte::core::CFBContainer::deleteFB(unsigned int const*&, unsigned int const*) at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/fbcontainer.cpp:130
0x420dd32b: CResource::executeMGMCommand(forte::core::SManagementCMD&) at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/resource.cpp:111 (discriminator 1)
0x420d91b5: CDevice::executeMGMCommand(forte::core::SManagementCMD&) at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/core/device.cpp:24
0x422add22: forte::ita::CommandParser::parseAndExecuteMGMCommand(char const*, char*) at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/stdfblib/ita/CommandParser.cpp:82
0x422ace75: DEV_MGR::executeRQST() at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/stdfblib/ita/DEV_MGR.cpp:92 (discriminator 3)
0x422acddd: DEV_MGR::executeEvent(unsigned int, CEventChainExecutionThread*) at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/stdfblib/ita/DEV_MGR.cpp:81
[... cut ...]
0x4201f195: CFreeRTOSThread::threadFunction(void*) at /home/franz/git/hr/LOGIBUS_integration_datapanel/4diac-forte/src/arch/freeRTOS/forte_thread.cpp:32

ELF file SHA256: 71563110f

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038050a
--- 0x4038050a: esp_cpu_wait_for_intr at /home/franz/esp-idf-v5.4/components/esp_hw_support/cpu.c:64

SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce2810,len:0x1658
load:0x403c8700,len:0x4
load:0x403c8704,len:0xe78
load:0x403cb700,len:0x2f7c
entry 0x403c8958
```

franz-hoepfinger-4diac, 2025-03-15:
processinterface.cpp:217
tested on different Projects. 
always same.

franz-hoepfinger-4diac, 2025-03-15:
https://docs.espressif.com/projects/esp-idf/en/v5.4/esp32s3/api-guides/fatal-errors.html

https://docs.espressif.com/projects/esp-idf/en/v5.4/esp32s3/api-guides/fatal-errors.html#instrfetchprohibited

InstrFetchProhibited

PC      : 0x00000000

franz-hoepfinger-4diac, 2025-03-15:
https://github.com/eclipse-4diac/4diac-forte/blob/bf2df4b592b3eaf7ef7b03f97146f972e2497104/src/core/io/processinterface.cpp#L214

franz-hoepfinger-4diac, 2025-03-16:
processinterface.cpp:39 --> Destructor Call.

franz-hoepfinger-4diac, 2025-03-16:
i made a Change: moved the deinitialise to a earlier Function, (EMGMCommandType::Kill)
and the Crash is gone. 

no i must check if all is Deinitialized correctly.

franz-hoepfinger-4diac, 2025-03-16:
I made a PR with this Changes: 

https://github.com/eclipse-4diac/4diac-forte/pull/363

diplfranzhoepfinger, 2025-03-16:
> My guess is I need to delete the handles.
> 
> A delete for the handle is nowhere called. Also the handle is not deregistered.
> 
> This I guess is the reason.

It is wrong. The Destruktor of the Handle is called; and there the Handle is deregistered. Automatically.

cochicde, 2025-03-16:
It seems that there are many calls to virtual functions in the destructor, which can be dangerous.

The inheritance of the class being destroyed is:
```
IX -> ProcessInterface -> IOObserver
                                         \-> CProcessInterfaceBase
```
1. ~IX finishes
2. ~ProcessInterface calls `ProcessInterface::deinitialise` (non virtual)
3. `ProcessInterface::deinitialise` calls `IOMapper::getInstance().deregisterObserver(this);`
4. `IOMapper::deregisterObserver` will call `dropHandle` (virtual from IOObserver) on `this` which is currently being destroyed  -> from a quick look, although risky it seems this still fine, since the `ProcessInterface` object is still being destroyed, so a pointer to it still does the dispatch correctly. However, the log says `non-virtual thunk to forte::core::io::ProcessInterface::dropHandle()` which seems that it can't find the `ProcessInterface::dropHandle` anymore
5. In any case, `ProcessInterface::dropHandle` will call then `QO()` and `STATUS()` which internally use `getDO` which is again a virtual function, in this case also overriden by IX (now not existing anymore). 

Maybe I missed something, and all calls to virtual functions are safe, but they seems risky. Wer could have `IOMapper::deregisterObserver` accept an extra parameter to drop the observer or not, so we can avoid the call back to the object being destroyed.

diplfranzhoepfinger, 2025-03-16:
the Access 
https://github.com/eclipse-4diac/4diac-forte/blob/bf2df4b592b3eaf7ef7b03f97146f972e2497104/src/core/io/processinterface.cpp#L214

does write to `QO()` in the Destructor. 
but destructors are called in reverse order. 
so the Element of the Subclass is no more existing, therefore the crash.

azoitl, 2025-03-16:
@cochicde while virtual methods may be dangerous in our case they are not the problem. But you have a point that we should be carfull here. A first step is to make `dropHandle` final which would give as more security. The other thing that I had in mind is to copy all I and Q blocks to the io folder and adjust them for the modular IO concept. This has the advantage that we do not have to take care about the old process interface and can fix this problem nicer.

As starting point I would merge PR #363 as it fixes @diplfranzhoepfinger immediate problem and work on from that.

azoitl, 2025-03-27:
With my work on cleaning up IO FBs (#365) this problem is now fully solved.

closed 2025-03-27 by azoitl