eclipse-4diac/4diac-forte #366: TODO: Test IX, QX if no INIT, and a REQ comes

franz-hoepfinger-4diac, 2025-03-17:
TODO: Test IX, QX if no INIT, and a REQ comes

we had recent crashes on FBs when they get a Event, but where not INIT. 

see also: 
https://github.com/espressif/esp-idf/issues/15586

@eliaseder thanks for reporting !

franz-hoepfinger-4diac, 2025-03-17:
@azoitl you can assign this to me.

azoitl, 2025-03-17:
@franz-hoepfinger-4diac please wait till my clean-up and rework of the I/O blocks is done (e.g., #365). I have more in the pipeline.

diplfranzhoepfinger, 2025-03-17:
> [@franz-hoepfinger-4diac](https://github.com/franz-hoepfinger-4diac) please wait till my clean-up and rework of the I/O blocks is done (e.g., [#365](https://github.com/eclipse-4diac/4diac-forte/pull/365)). I have more in the pipeline.

exactly this i want to do. 
therefore i wrote it down here.