eclipse-4diac/4diac-forte #379: [CI] Formatting check

cochicde, 2025-03-24:
We need a formatting check added to the CI to have the code always consistent. We have the `[fortestyle.xml](https://github.com/eclipse-4diac/4diac-forte/blob/release/fortestyle.xml)` which are derived from what's specified here [forte_coding_rules.tex](https://github.com/eclipse-4diac/4diac-forte/blob/release/doc/coding_rules/src/forte_coding_rules.tex).

I'm not sure if we can incorporate a formatting check based on `fortestyle.xml`, and if not, we could add the same rules in clang which I know is easier to incorporate to the CI.

Any suggestions/ideas are welcomed

mx990, 2025-03-25:
I would prefer `.clang-format` in general, since that also works with other IDEs, such as CLion or Visual Studio (Code). It seems this is also used by Eclipse CDT itself, see https://github.com/eclipse-cdt/cdt/blob/main/.clang-format.

azoitl, 2025-03-25:
In general I'm open to anything. But I didn't find an option to use `.clang-format` in CDT. Maybe I have to check my setup.

MartinMelikMerkumians, 2025-03-26:
> In general I'm open to anything. But I didn't find an option to use `.clang-format` in CDT. Maybe I have to check my setup.

I think you need a plugin for that https://marketplace.eclipse.org/free-tagging/clang-format

azoitl, 2025-03-26:
@MartinMelikMerkumians I will also recheck the latest CDT 12.0 and install it from scratch. There is now a clangd integrated. I guess for that I can use a clang format file.

azoitl, 2025-03-26:
I've tested with ned CDT 12.0 and its LSP based setup. There I can us a clang format file. So I'm all in. How do we get such a file?

mx990, 2025-03-26:
I will create a `.clang-format` file based on the coding rules.

azoitl, 2025-03-26:
> I will create a `.clang-format` file based on the coding rules.

Thanks a lot. Not sure if I'm opening a pandoras box but if we have rules that are by now outdated or do not make to much sense anymore I would consider changing them. @MartinMelikMerkumians  what is your opinion here?

mx990, 2025-03-26:
I have created a PR https://github.com/eclipse-4diac/4diac-forte/pull/389 with a `.clang-format` file based on the coding rules and what I would consider sane defaults.

gravemalte, 2025-06-26:
I think we can close this issue. DO you agree @cochicde and @mx990?