eclipse-4diac/4diac-forte #380: Copying data from CIEC_ARRAY_FIXED is a shallow copy

gravemalte, 2025-03-24:
First of all, we're not sure if we're just not smart enough to use the API right, or if something is broken.

@m-meingast and I tried to read data from an Adapter connection using a Function Block.
We used the Adapter (auto-generated) method `var_INDD()` to get the data.

The data in this case has the `CIEC_ARRAY_FIXED<CIEC_REAL, 0, 499>` type (the boxed IEC type shouldn't matter).

We would suspect when we want to get the data, we can copy (deep copy) it to a buffer we would use

`CIEC_ARRAY_FIXED<CIEC_REAL, 0, 499> data = var_Socket.var_INDD()`
or
`CIEC_ARRAY_FIXED<CIEC_REAL, 0, 499> data(var_Socket.var_INDD())`

But this doesn't work. Only when we are explicitly stating that the data is a reference,
like:
`CIEC_ARRAY_FIXED<CIEC_REAL, 0, 499> &data = var_Socket.var_INDD()`
or
`CIEC_ARRAY_FIXED<CIEC_REAL, 0, 499> &data(var_Socket.var_INDD())`

We would suspect that the copy constructor/assignment operator would create a deep copy of the array and returns it. 
But as far as we understand, we only get a shallow copy. Is this intended?

MartinMelikMerkumians, 2025-03-24:
Hi,

let me rephrase your issue.
you are saying that neither

`CIEC_ARRAY_FIXED<CIEC_REAL, 0, 499> data = var_Socket.var_INDD()`
nor
`CIEC_ARRAY_FIXED<CIEC_REAL, 0, 499> data(var_Socket.var_INDD())`

is creating a deep copy of the array?
Can you post the declaration of `var_Socket.var_INDD()`

And, if my C++ is not massively rusted in, neither
`CIEC_ARRAY_FIXED<CIEC_REAL, 0, 499> &data = var_Socket.var_INDD()`
nor
`CIEC_ARRAY_FIXED<CIEC_REAL, 0, 499> &data(var_Socket.var_INDD())`
is creating a copy, but only binds the reference of `var_Socket.var_INDD()` to `data`

Another question, what is your goal/scenario here? You wanted a copy and you didnt get one? Because this is checked in test cases (https://github.com/eclipse-4diac/4diac-forte/blob/develop/tests/core/datatypes/CIEC_ARRAY_test.cpp) and should work for all kinds of combinations.
Additionally, you would see the original values even in the case of a shallow copy, as shallow copy would mean the reference is copied instead of a new object is created (which is the case with your "solution" on a high level)

If your scenario is, you modified someting in data and had the same changes in the original data source, then it would be some kind of shallow copy, and I admit that this is not covered by the test cases, but could easily added by you to verify your issue.

Checking the code, I do not think that the problem is in `CIEC_ARRAY_FIXED`, as it calls the copy ctor of std::array, and as `data` stores `CIEC_REAL` objects, it calls its ctor, which copies the interally stored value.

Tl:dr: So without seeing anything of your code, my first guess would be that there is something wrong in your code.

gravemalte, 2025-03-24:
Hi!

Okay, maybe I didn’t explain this clearly. We spent several hours troubleshooting an issue where we were receiving random data from the adapter connection.

Our goal is to receive data from the adapter, load it into a buffer, and then manipulate it and write it into the plug. Our idea was to take the incoming data from the adapter and create a deep copy of it (of the `std::array data` member).

We never encountered a type error because, if we didn’t bind the result to a reference, we would get a something different instead (a pointer?).
I know that the type deduction is not able to deduct reference bindings. But I thought you can overload the `=` so only a reference assignment is valid and not the copy assignment.  Or am I getting something mixed up?

Our code looked like this:

```cpp
void FORTE_LowPassFilter::alg_setData(void) {
  auto data(var_Socket.var_INDD());
  std::vector<CIEC_REAL> buffer(500);

  auto j = 0;
  for(int i = 0; i < data.size(); ++i){
    if(data[i] < 50000){
      buffer[j++] = data[i];
    }
  }
  for(int i = 0; i < data.size(); ++i){
    var_PLUG1.var_INDD()[i] = buffer[i];
  }
}
```

After 2.5 hours of debugging, we found the issue was the missing reference binding that you mentioned.

m-meingast, 2025-03-24:
I may also add to that: During debugging we noticed that using `CIEC_ARRAY_FIXED<CIEC_REAL, 0, 499> data = var_Input.var_INDD();` leads to the call of the copy constructor 
```cpp
CIEC_ARRAY_FIXED(const CIEC_ARRAY_FIXED &paSource) : data(paSource.data) {}
```

MartinMelikMerkumians, 2025-03-25:
This test case:

```
BOOST_AUTO_TEST_CASE(StaticToStaticCopyConstructor)
{
  CIEC_ARRAY_FIXED<CIEC_INT, 0, 10> intArraySource = {CIEC_INT(1), CIEC_INT(2), CIEC_INT(3), CIEC_INT(4),
                                                   CIEC_INT(5), CIEC_INT(6), CIEC_INT(7), CIEC_INT(8),
                                                   CIEC_INT(9), CIEC_INT(10), CIEC_INT(11)};

  CIEC_ARRAY_FIXED<CIEC_INT, 0, 10> intArray(intArraySource);

  BOOST_CHECK_EQUAL(static_cast<CIEC_INT::TValueType>(intArray[0]), 1);
  BOOST_CHECK_EQUAL(static_cast<CIEC_INT::TValueType>(intArray[1]), 2);
  BOOST_CHECK_EQUAL(static_cast<CIEC_INT::TValueType>(intArray[2]), 3);
  BOOST_CHECK_EQUAL(static_cast<CIEC_INT::TValueType>(intArray[3]), 4);
  BOOST_CHECK_EQUAL(static_cast<CIEC_INT::TValueType>(intArray[4]), 5);
  BOOST_CHECK_EQUAL(static_cast<CIEC_INT::TValueType>(intArray[5]), 6);
  BOOST_CHECK_EQUAL(static_cast<CIEC_INT::TValueType>(intArray[6]), 7);
  BOOST_CHECK_EQUAL(static_cast<CIEC_INT::TValueType>(intArray[7]), 8);
  BOOST_CHECK_EQUAL(static_cast<CIEC_INT::TValueType>(intArray[8]), 9);
  BOOST_CHECK_EQUAL(static_cast<CIEC_INT::TValueType>(intArray[9]), 10);
  BOOST_CHECK_EQUAL(static_cast<CIEC_INT::TValueType>(intArray[10]), 11);
}
```

should cover your use case and is tested, so the copying should work. So from my point of view, whatever you are experiencing its not an issue with the copy ctor or copy assignment, but somehow how you use the adapter variable or an issue with adapters itself.

For further help or analysis the definition of your adapter would be of help, otherwise I do not see how this can be further inspected.

On a different note, when this was about the code you where using, why do you need an intermediate buffer?

```
void FORTE_LowPassFilter::alg_setData(void) {
  auto data(var_Socket.var_INDD());
  std::vector<CIEC_REAL> buffer(500);

  auto j = 0;
  for(int i = 0; i < data.size(); ++i){
    if(data[i] < 50000){
      buffer[j++] = data[i];
    }
  }
  for(int i = 0; i < data.size(); ++i){
    var_PLUG1.var_INDD()[i] = buffer[i];
  }
}
```

under the assumption that everything would work as it should and `var_Socket.var_INDD()` is an array of type `CIEC_REAL` and sized from 0 to 500

```
void FORTE_LowPassFilter::alg_setData(void) {

  auto j = 0;
  for(int i = 0; i < var_Socket.var_INDD().size(); ++i){
    if(var_Socket.var_INDD()[i] < 50000){
     var_PLUG1.var_INDD()[i][j++] = var_Socket.var_INDD()[i];
    }
  }
}
```
 should also work (if I did not overlook something). 

Perhaps @mx990 or @azoitl can see something I overlook in this issue?

mx990, 2025-03-25:
It is very difficult to understand the actual problem from your description. Maybe @gravemalte or @m-meingast could again state the expected vs. the current behavior more clearly (incl. actual values) and also provide a minimal (non-)working example to go from?

gravemalte, 2025-03-26:
So, we wanted to get an array via an adapter and send an array via an adapter. We then wanted to create a copy of this array. Why a copy? Because we didn't want to work on the original data. Imagine having a function block with two adapter outputs that should send different data. Adapter connection 1 should send the original data, while adapter connection 2 should forward the manipulated data. This is a simplified example explaining why we want a deep copy of the array rather than working on the original reference.

However, the issue is that with the statement `auto data(var_Socket.var_INDD());`, we would expect to receive a copy of the array via the adapter. But that’s not the case. Instead, we seem to be getting memory addresses. This behavior is not what I would expect from this code.

I created a Gist with the example code, you can find it here: https://gist.github.com/gravemalte/09a399f42283186259c00e275457554b

azoitl, 2025-03-26:
After looking into the problem I saw that as @MartinMelikMerkumians stated the problem is not in the copying of the array. The problem is in the code generated for Adapter Interface blocks. There the accessor methods for inputs and outputs are using the `CIEC_ARRAY_FIXED` however the base class is creating arrays using the `CIEC_ARRAY_DYNAMIC`. So a quick fix would be to change the generator for adapter interfaces. But I think this would be a bit short sighted. We had somewhere already a discussion that the current adapter interface implementation with exchanging pointers between plug and sockets. 

@mx990 @MartinMelikMerkumians what do you think?

@gravemalte can you for testing manually change the code for your adapter interface blocks.

mx990, 2025-03-26:
I agree with your assessment of the situation. I believe you are referring to the discussion in https://github.com/eclipse-4diac/4diac-forte/pull/301 and the following issue https://github.com/eclipse-4diac/4diac-forte/issues/306. I also believe that adapters are currently somewhat problematic in general and that we should work on a permanent solution soon.

MartinMelikMerkumians, 2025-03-26:
As stated in our chat @azoitl I also agree with your assessment, and as it seem to work for @gravemalte now, this seems to be the base issue here. I would therefore propose to close this issue as solved now.