eclipse-4diac/4diac-forte #388: Compile error in default cmake config setup, due to E_N_TABLE is dependent on F_SUB

MartinMelikMerkumians, 2025-03-26:
After creating a default CMake config on Linux for posix on my WSL Debian and trying to build it, i get

```
In file included from /home/mmm/4diac-forte/src/stdfblib/events/E_N_TABLE_fbt.cpp:19:
/home/mmm/4diac-forte/src/stdfblib/events/E_N_TABLE_fbt.h:33:10: fatal error: F_SUB_fbt.h: No such file or directory
   33 | #include "F_SUB_fbt.h"
```
F_SUB is part of the IEC 61131-3 library which is disabled by default, which leads to this error.

As a fix, i propose to add the IEC 61131-3 lib at default and make the EVENTS lib dependent on IEC 61131-3 in CMake (I think at least this should be possible by activating the IEC 61131-3 flag on loading events)

If that's not desirable, then the dependence needs to be removed from E_N_TABLE

Please also find attacked my CMakeCache.txt for reference

[CMakeCache.txt](https://github.com/user-attachments/files/19468772/CMakeCache.txt)

MartinMelikMerkumians, 2025-03-26:
Do you mean FBs with add 1 or subtract 1 from an input IN? so essentially

`OUT := IN + 1;`
and
`OUT := IN - 1;`
?

azoitl, 2025-03-26:
> https://www.holobloc.com/doc/navigator.htm
> 
> https://www.holobloc.com/doc/fb/rt/events/E_N_TABLE.htm
> 
> We could make 2 blocks E_INCR E_DECR which Outputs IN+1 and IN-1 which could replace this -1 Block.
> 
> These 2 blocks would be useful in other cases too.

@diplfranzhoepfinger So you mean an E_CTU and an E_CTD?

cochicde, 2025-03-29:
In the `utils` module, `FORTE_E_BLINK` and and `FORTE_E_BLINK_TRAIN` have a dependency on `GEN_ADD_fbt` also from the IEC 61131 module.

I think setting the IEC61131 module as default is too strong. The problem here is the dependency between modules.  A CMake error should be enough if the dependency is not removed.

diplfranzhoepfinger, 2025-03-29:
> [@diplfranzhoepfinger](https://github.com/diplfranzhoepfinger) So you mean an E_CTU and an E_CTD?
no, just a function which Outputs IN+1 and IN-1 like the C Operators i++ and i--
then can be used as a Counter as well, if you loop back the OUT to the IN.

E_CTU and E_CTD is different, as they are FunctionBlock with a internal State. 

you could perfectly Teach then the Difference between Function and Function Block,

(the E_N_TABLE  does not really need SUB, it only needs a offset of 1->0; 2->1, 3->2, 4->3 per Standard.

MartinMelikMerkumians, 2025-03-30:
> In the `utils` module, `FORTE_E_BLINK` and and `FORTE_E_BLINK_TRAIN` have a dependency on `GEN_ADD_fbt` also from the IEC 61131 module.
>
> I think setting the IEC61131 module as default is too strong. [...]

I agree, that the whole 61131 lib could be too much, but the base operators could (and should imho)  be included as default.

MartinMelikMerkumians, 2025-03-30:
> E_CTU and E_CTD is different, as they are FunctionBlock with a internal State.
>
> you could perfectly Teach then the Difference between Function and Function Block,
>
> (the E_N_TABLE does not really [...]

I don't think adding a decrenemt and increment function is the right solution, as it simply duplicates the + and - operator with one fixed operand.

The benefit it gives in C/C++/Java and wherever it exists is to give you a textual shortcut.

Also we should not design our standard library based on educational benefits.

diplfranzhoepfinger, 2025-03-30:
> I agree, that the whole 61131 lib could be too much, but the base operators could (and should imho) be included as default.
that would be another Idea. 

to say math (SUB, ADD, MUL) is seperated from 61131, and a smaller lib. 
many people think the 4d 61499 lib is heavy. 

this might solve 2 Problems then.

diplfranzhoepfinger, 2025-03-30:
> The benefit it gives in C/C++/Java and wherever it exists is to give you a textual shortcut.
> 
> Also we should not design our standard library based on educational benefits.

agree to that.