eclipse-4diac/4diac-forte #392: [Proposal] Rework FB Interface Spec Data Handling

azoitl, 2025-03-27:
We are currently working on the final clean-ups and API breaking changes for the Eclipse 4diac 3.0 release. As there are anyhow major changes for 4diac FORTE happening I was reconsidering the way how we store the data for FB interface specs. As we have here many raw pointers this was always a place which I was not happy about. 

The first proposal I have is to store all the arrays with the interface data in `std::array`s and move them out of the class so that they are only in the cpp file of a FB. For a FB (I took `E_CTU` as example) this would then look like as follows:

```
namespace FORTE_E_CTU_data {
  const auto cEventInputNames    = std::array{STRID(CU), STRID(R)};
  const auto cEventInputTypeIds  = std::array{STRID(Event), STRID(Event)};
  const auto cEIWith             = std::array<const TDataIOID, 2>{0, CFunctionBlock::scmWithListDelimiter};
  const auto cEIWithIndexes      = std::array<const TForteInt16, 2>{0, -1};
  const auto cEventOutputNames   = std::array{STRID(CUO), STRID(RO)};
  const auto cEventOutputTypeIds = std::array{STRID(Event), STRID(Event)};
  const auto cEOWith             = std::array<const TDataIOID, 6>{0, 1, CFunctionBlock::scmWithListDelimiter, 0, 1, CFunctionBlock::scmWithListDelimiter};
  const auto cEOWithIndexes      = std::array<const TForteInt16, 2>{0, 3};
  const auto cDataInputNames     = std::array{STRID(PV)};
  const auto cDataInputTypeIds   = std::array{STRID(UINT)};
  const auto cDataOutputNames    = std::array{STRID(Q), STRID(CV)};
  const auto cDataOutputTypeIds  = std::array{STRID(BOOL), STRID(UINT)};
}
```

This change would still be fully backwards compatible and could just be done in the exporter. The initialisation of the interface spec struct would then be:

```
const SFBInterfaceSpec FORTE_E_CTU::scmFBInterfaceSpec = {
  FORTE_E_CTU_data::cEventInputNames.size(), FORTE_E_CTU_data::cEventInputNames.data(), FORTE_E_CTU_data::cEventInputTypeIds.data(), FORTE_E_CTU_data::cEIWith.data(), FORTE_E_CTU_data::cEIWithIndexes.data(),
  FORTE_E_CTU_data::cEventOutputNames.size(), FORTE_E_CTU_data::cEventOutputNames.data(), FORTE_E_CTU_data::cEventOutputTypeIds.data(), FORTE_E_CTU_data::cEOWith.data(), FORTE_E_CTU_data::cEOWithIndexes.data(),
  FORTE_E_CTU_data::cDataInputNames.size(), FORTE_E_CTU_data::cDataInputNames.data(), FORTE_E_CTU_data::cDataInputTypeIds.data(),
  FORTE_E_CTU_data::cDataOutputNames.size(), FORTE_E_CTU_data::cDataOutputNames.data(), FORTE_E_CTU_data::cDataOutputTypeIds.data(),
  0, nullptr,
  0, nullptr
};
```

However I'm also not really happy with that and this brings to the core of writing this issue. I would also like to rework `SFBInterfaceSpec` as follows:

```
struct SNewFBInterfaceSpec {
  const std::span<const CStringDictionary::TStringId> mEINames;
  const std::span<const CStringDictionary::TStringId> mEITypeNames;
  const std::span<const TDataIOID>                    mEIWith;
  const std::span<const TForteInt16>                  mEIWithIndexes;
  const std::span<const CStringDictionary::TStringId> mEONames;
  const std::span<const CStringDictionary::TStringId> mEOTypeNames;
  const std::span<const TDataIOID>                    mEOWith;
  const std::span<const TForteInt16>                  mEOWithIndexes;
  const std::span<const CStringDictionary::TStringId> mDINames;
  const std::span<const CStringDictionary::TStringId> mDIDataTypeNames;
  const std::span<const CStringDictionary::TStringId> mDONames;
  const std::span<const CStringDictionary::TStringId> mDODataTypeNames;
  const std::span<const CStringDictionary::TStringId> mDIONames;
  const std::span<const SAdapterInstanceDef>          mAdapterInstanceDefinition;
    
  TEventID getNumEIs() { 
    return mEINames.size();
  }
  
  TEventID getNumEOs() { 
    return mEONames.size();
  }
  
  TPortId getNumDIs() { 
    return mDINames.size();
  }
  
  TPortId getNumDOs() { 
    return mDONames.size();
  }
  
  TPortId getNumDIOs() { 
    return mDIONames.size();
  }
  
  TPortId getNumAdapters() { 
    return mAdapterInstanceDefinition.size();
  }
};
```

For this we need the `std::span` class introduced and recommended by the [C++ Core Guidelines](https://github.com/isocpp/CppCoreGuidelines].  However to get this class we either a new dependency to [GSL: Guidelines Support Library](https://github.com/microsoft/GSL) or we need to upgrade to C++20. Both have their pros and cons.

Also I wanted to put that issue here to see if I'm not totally of track and there is even a better solution for this problem.

azoitl, 2025-03-27:
I checked a bit what is the minimum GCC version that provides `std::span` and this is GCC 10. We anyhow need at least GCC 11.3 because of a bug in `std::variant` so maybe changing to C++20 is not that big deal that I thought.

MartinMelikMerkumians, 2025-03-28:
> I checked a bit what is the minimum GCC version that provides `std::span` and this is GCC 10. We anyhow need at least GCC 11.3 because of a bug in `std::variant` so maybe changing to C++20 is not that big deal that I thought.

First, I want to say I like the proposal and would opt to go for it, and I would also propose to advance in the C++ version, as we would have liked several features already some time ago.

Still, for the second point, that its perhaps not that a hassle to change to C++20 because std::span is already available in GCC 10, I need to say that this is perhaps a bit too much of a simplification.
Certain C++XX features where always available in earlier versions in certain compilers, but agreeing on using C++20 would imply that ALL of those features are usable in the codebase.

azoitl, 2025-03-28:
> We have to check if std::span works correctly in Visual Studio 2022 compiler. 
A quick check on the Visual Studio web page states that `std::span` is supported since VS 2019

> What is the impact of all this changes in creating interface decriptions for dynamic function blocks?

It will greatly simplify the data management there as well. You can put the data in `std::vector` and let vector manage the data. But it is also possible that you us plain arrays (do not recommend). But espeically with setup is simpler.

cochicde, 2025-03-28:
it seems that the charactistics of `span` are exactly what's needed for the interface. `SPinDescriptor` and `SEventPinDescriptor` makes sense also to me. 

Regarding how to have it in the code, I'm not sure if it's a good idea to make the jump to C++20, since you loose some usability with it. That will not only break API, but make some targets not supported anymore. In this particular case, on a first impression, it seems that it might be a big price to pay, for "just" one part of the code. At the end, what's needed here is not `std::span`, but a an abstraction that offers a "view" of an array and its size, and offers iteration capabilities. It might be a good idea to ask about C++20 support in the comming 4diac community meeting to have a better overview of the impact.

If using GLS or any other external implementation is not a big issue regarding license or setup, I think is the best option since you get all the benefits and none of the disadvantages.

gravemalte, 2025-03-28:
Also, my 2 cents.

I like the API change a lot, also the structs are a good fit.

From a 4diac-FBE point of view (at least for gcc of the linux targets) C++20 would be supported. We have a gcc 13.2.0 for linux targets. @jwalt, I'm not sure about the mingw toolchains, maybe you can say something about that.

Regarding GLS: It's more a personal preference, but I would not add any further dependencies. FORTE depends on the core base on hardly any dependencies, and that's what makes it so lean.
This would create a dependency that (as I see it) would be inescapable and destroy the idea of ‘I can compile it with any compiler in a basic form’.
Licence technically it may not be a problem, GSL is licensed under an MIT licence, but I wouldn't like to add core dependency.

I would prefer C++20 or even our own span template implementation.

mx990, 2025-03-30:
I also really like the proposed change, particularly the `std::span` and the structs.

I would suggest to use an [unnamed namespace](https://en.cppreference.com/w/cpp/language/namespace#Unnamed_namespaces) for the arrays, which would also give them internal linkage. Considering the number of FBs in forte, this could also significantly reduce the number of entries in the symbol table and thus both binary size and link times.

Regarding the `std::span`, I would also prefer C++20 or our own implementation, instead of adding an additional dependency.

With C++20, we could further use [designated initializers](https://en.cppreference.com/w/cpp/language/aggregate_initialization#Designated_initializers), in order to be more robust against future changes or extensions:
```cpp
const SNewFBInterfaceSpec2 E_CTU_scmFBInterfaceSpec2 = {
  .mEventInputs = FORTE_E_CTU_data::cEventInputs,
  .mEventOutputs = FORTE_E_CTU_data::cEventOutputs,
  .mDataInputs = FORTE_E_CTU_data::cDataInputs,
  .mDataOutputs = FORTE_E_CTU_data::cDataOutputs
};
```