eclipse-4diac/4diac-forte #394: [CAN Layer] Unified CAN Layer with Modular Design

gravemalte, 2025-03-30:
I am currently planning an implementation of a CAN layer in 4diac-FORTE. A prototype layer can be found in PR #367. However, this is purely experimental and currently only supports Linux with SocketCAN. Additionally, data handling in this implementation is quite static, which is not optimal.

I have given this some thought and, with the help of @azoitl, gathered ideas on how the final implementation could look.

At the moment, I see two possible scenarios:
## 1. A "Raw" Communication Layer

This layer is responsible for forwarding raw data to the appropriate target. Additional layers can be added on top to enable data transformations, similar to how it is handled in the MQTT layer.

By "raw" layer, I mean that data is sent directly to a specific device, file descriptor (FD), or another resource, without any additional abstraction, such as specific CAN specifications (e.g., CiA 301).

An example component diagram illustrating this concept can be seen here:

[image]

## 2. A Complete CAN Abstraction Layer
This approach would introduce a fully-fledged CAN abstraction layer, similar to how the OPC UA layer is 
structured in 4diac-FORTE. For this, CANopenNode can be utilized [[0]](https://github.com/CANopenNode/CANopenNode).

CANopenNode is an open-source CANopen stack implementing the CiA 301 standard. It provides functionality for PDO (Process Data Objects), SDO (Service Data Objects), NMT (Network Management), EMCY (Emergency Messages), and other core CANopen features. CANopenNode provideds a abstraction layer for RTOS'es and Linux (so no need to communicate with the drivers directly) [[1]](https://github.com/CANopenNode/CANopenLinux),  [[2]](https://docs.zephyrproject.org/latest/samples/modules/canopennode/README.html), [[3]](https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/modules/canopennode/src/main.c#L210).

### Advantages of a CANopen Abstraction Layer:

1. Standardized Communication: CANopen ensures compatibility across different devices and manufacturers.
2. Modular and Extensible: The abstraction allows for seamless integration of higher-level communication protocols.
3. Reduced Development Effort: Developers can interact with the CAN bus through a well-defined API instead of handling raw CAN frames.
4. Network Management Features: CANopen includes built-in support for device discovery, heartbeat monitoring, and error handling.

A component diagram can be seen here:

[image]

## Common Base for CAN Communication

Since both approaches (Raw CAN + CANopen) should be available in 4diac-FORTE, I plan to develop a common base layer that will serve as the foundation for both implementations. This will ensure that both solutions share:

1. A unified interface for CAN communication
2. Flexible configuration options
3. A scalable architecture that can support additional features in the future

## Next Steps

I will continue developing this common CAN communication layer, while also prototyping an IEC 61499 application to validate my ideas and ensure they work in real-world scenarios.

I would appreciate any feedback, suggestions, or ideas that could improve this approach. Let me know what you think!

cochicde, 2025-03-31:
Something kind of similar to the abstraction layer in the `raw` method is done in `CSerialComLayerBase` which tries to abstract the serial communication to both linux and windows.

I don't know much about CAN standards, but it seems to me that both methods presented here are different layers using the same lower socket configuration. In the raw version, you send whatever you want (types are not defined) using the CAN socket. 

The second method sounds like what we have in OPCUA now, where we use the layer also to map to the types and objects defined in the standard. In this case, I assume you will need somehow to define these PDO, SDO, ... in 61499 (via PARAMS, specialized FBs, ...) which will then connect to the `CANopenNode` API.

gravemalte, 2025-03-31:
Exactly that, because I have considered the `CSerialComLayerBase` as the base for the plan. But I don't like the idea of storing the specialization implementation of it in the platform directory for each architecture. Maybe I come up with another idea.

And I was also inspired by the OPC UA layer for the second one, basically the same thing is done there, you can definitely see parallels here. However, I think it will get a bit tricky when you deal with RTOS platforms, because we will need to provide a good abstraction there.

franz-ms-muc, 2025-05-05:
for "2": 

mainly the CAN Stacks are bound more tightly to the Hardware. 
you might also want to run these Stacks outside of 4diac. 

e.g. in our Implmentation we have the HMI (DefaultPool.iop) on the Webserver parallel to the *.fboot, 
as we want the HMI to be shown on the Display independent of the 4diac FORTE. 

happy to show you our Solution in a real-world demonstration in Munich or in Linz. 

Tnx, 
Franz