eclipse-4diac/4diac-forte #399: Upgrade 4diac FORTE to C++ 20

azoitl, 2025-04-04:
In our work towards the Eclipe 4diac 3.0 release with many major changes and major improvements we noticed that for implementing 4diac FORTE more correct and saver an upgrade to C++ 20 would be beneficial. In recent discussion we, for example, identified the following C++20 features:

- virtual constexpr methods and destructors, particularly for CIEC datatypes
  - enables constexpr literal operators for CIEC datatypes
  - enables usage of CIEC datatypes in switch labels, see https://github.com/eclipse-4diac/4diac-ide/issues/1106#issuecomment-2683165360
- literal operators for string ids, see https://github.com/eclipse-4diac/4diac-forte/pull/338#issuecomment-2691184906
- designated initializers, particularly for interface spec, see https://github.com/eclipse-4diac/4diac-forte/issues/392#issuecomment-2764716525
- concepts to make templates more type-safe, particularly for CIEC datatypes and 61131 functions
- aliasing shared_ptr construktor, see (8) in https://en.cppreference.com/w/cpp/memory/shared_ptr/shared_ptr, for better handling of connections
- std::span for improved interface spec handling: https://github.com/eclipse-4diac/4diac-forte/issues/392

And I guess there several more features that would be helpful.

Therefore I wanted to propose to upgrade the minimum required C++ version for 4diac FORTE to C++ 20.

This has some implication as maybe more exotic devices do not have an up to date compiler.

Therefore with this issue I want to assess explicitly with our users what compiler versions they use and what limitations are out there.

diplfranzhoepfinger, 2025-04-04:
Thanks for bring this up, i asked https://github.com/espressif if there are any issues.

ruspl-afed, 2025-04-04:
Perhaps it would be helpful to specify what kind of migration effort is expected from suppliers

azoitl, 2025-04-04:
> Perhaps it would be helpful to specify what kind of migration effort is expected from suppliers

@ruspl-afed the migration effort will be to use a compiler version that provides the according features. For GCC it would be somthing around v 12 or 14 (see https://en.cppreference.com/w/cpp/compiler_support/20 for details). If you use 4diac FBE no further steps should be needed. You would see it rather quick as 4diac FORTE would not compile with a wrong compiler.

jwalt, 2025-04-04:
I think this needs highlighting:

Users on old development platforms must switch to FBE or some other way to use a recent compiler. There are Linux distributions that are still supported but that will no longer be able to compile FORTE with the system default compiler. This includes Linux on CI servers, if your workflow relies on that.

Also, if you absolutely rely on MSVC, FBE can't help you (FBE uses mingw for windows). In that case you must update to a decently recent version (see Alois' link for details).

mx990, 2025-04-04:
I believe this would result in the following minimum supported compiler versions[^1]:
- gcc 12
- clang 19
- MSVC 19.42 / VS 2022 17.12

This excludes some features that should not be relevant for FORTE but would otherwise require more recent versions.

diplfranzhoepfinger, 2025-04-04:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/cplusplus.html#c-language-standard

so for us, even C++23 with Gnu Extensions would be OK!

diplfranzhoepfinger, 2025-04-04:
> Perhaps it would be helpful to specify what kind of migration effort is expected from suppliers

if you are on Forte 2.0.1 the Migration is anyhow heavy, so this does not really add a lot.

oellith, 2025-04-07:
We are currently using:

- Visual Studio 2022 Version 17.9.6
- gcc 12.2

azoitl, 2025-04-16:
As we mostly got positive feedback and nobody raised any show stoppers we would merge the dependency change on Friday April 18th. 

So this is the last chance to send us vetos.

azoitl, 2025-04-18:
Just merged PR #404. With that we now are requiring C++ 20.

Thx to everybody for the discussion and feedback.

closed 2025-04-18 by azoitl