eclipse-4diac/4diac-forte #410: Timer function block jitter

nojapower, 2025-04-16:
Hi,

This relates to the following:
https://www.eclipse.org/forums/index.php/t/1105240/
https://bugs.eclipse.org/bugs/show_bug.cgi?id=568902

We tested with 4diac FORTE 2.0.1 and found an improvement, however, the issue persists. After looking at the code we found a solution we hope might help.

This change to "forte/src/arch/posix/pctimeha.cpp" in CPCTimerHandler::run() resolves the jitter issue.
```
void CPCTimerHandler::run(){
  /* Set period for tick */
  struct timespec stPeriod;
  stPeriod.tv_sec = 0;
  stPeriod.tv_nsec = (1000000 / getTicksPerSecond()) * 1000;

  /* Next tick time is always:
   *     stTimeNext = stTimeNext + stPeriod */
  struct timespec stTimeNext;
  clock_gettime(CLOCK_MONOTONIC, &stTimeNext);

  while ( isAlive() ) {

    /* stTimeNext = stTimeNext + stPeriod */
    timespecAdd(&stTimeNext, &stPeriod, &stTimeNext);

    /* Wait till stTimeNext before we tick or get a signal to shutdown */
    clock_nanosleep(CLOCK_MONOTONIC, TIMER_ABSTIME, &stTimeNext, NULL);

    /* Tick every stPeriod seconds */
    nextTick();

  }
}
```

There is a subsequent issue that setting 1ms as the tick period results in a 2ms tick. This change to "forte/src/arch/timerha.cpp" in CTimerHandler::nextTick(void) resolves the issue.
```
void CTimerHandler::nextTick(void) {
  ++mForteTime;
  mDeviceExecution.notifyTime(mForteTime); //notify the device execution that one tick passed by.

  if(!mRemoveFBList.empty()){
    processRemoveList();
  }

  if(0 != mAddFBList){
    processAddList();
  }

  /* NOJA - processTimedFBList() must be after processAddList() */
  processTimedFBList();
}
```

Thank you

azoitl, 2025-04-16:
@nojapower many thx for this feedback and suggestion. Especially the one for "forte/src/arch/posix/pctimeha.cpp" is a very good idea. Could you please turn this into a PR.

Regarding the "forte/src/arch/timerha.cpp" I had a very specific reason for doing it in the existing order (not that I remember it any more). However looking at our code I noticed that maybe this could even be better fixed by changing the following line in processAddList:

 `if(entry.mTimeOut < mForteTime) {`

to 

 `if(entry.mTimeOut <= mForteTime) {`

Because then the FB wouldn't even inserted into our timer list. Or did I miss anything?

nojapower, 2025-04-16:
Hi @azoitl

Yes, your change to "forte/src/arch/timerha.cpp" in CTimerHandler::processAddList() resolves the missed tick issue and looks like a better solution.
```
if(entry.mTimeOut <= mForteTime) {
```

I looked at the contribution guide but I'm not confident with the information in the instructions and translating them into git command line operations. We use git here and if its anything like the bad experience we have here with git I might end up deleting your repository along with half the internet. I hope it is ok if I pass on tinkering with your git instance.

Thank you

azoitl, 2025-04-17:
I understand that git command line can be intimidating.  I'm also not a big fan of it and avoid it as much as possible. For my daily needs I only use the git functionality offered by the Eclipse IDE. This makes it for me much saver and faster to handle.

Regarding your request on doing the changes on your behave. I can do that for the change I proposed, but I fear this will not be possible for the your proposed change. Although it is small it is already something that has its individual copyright. So I would at least need you to sign up at Eclipse Foundation and sign the Eclipse Contributor Agreement. Please note that you need to do that as individual person not with a company account. Otherwise we may legally compromise  4diac FORTE and I'm not willing to do that.

I'm happy to set up an online session to support you in the process. Especially contributing via Github got much easier and saver for both side.

azoitl, 2025-04-17:
The processAddList fix is already in: #416