eclipse-4diac/4diac-forte #423: forte freeze build error:UA_ServerConfig has no member named logging.

wzj7531, 2025-04-28:
Built forte of freeze version and then an error: UA_ServerConfig has no member named 'logging'. Is it open62541 version problem?

azoitl, 2025-04-28:
4diac FORTE develop and freeze version require a newer version of open62541. Currently this is version 1.4.7. 

You can always check which open62541 version we use in the file `dependencies.spdx` which you can find in the 4diac FORTE's root directory. 

You may also want to look into [4diac FBE](https://github.com/eclipse-4diac/4diac-fbe) which is, among other things, automating the dependency handling of 4diac FORTE.

wzj7531, 2025-04-29:
I originally used Forte FBE to build it, but this error still occurs. Is it because the Forte FBE version requires the use of develop instead of release?

azoitl, 2025-04-29:
yes 4diac FBE requires a develop version. If you take the zip file bundled with the alpha release of 4diac FBE: https://github.com/eclipse-4diac/4diac-fbe/releases/tag/beta-2025-02 and follow the documentation in the readme file there it should be possible to build a correct version of 4diac FORTE.

closed 2025-05-22 by azoitl