eclipse-4diac/4diac-forte #461: Incorrect character value for $L escape sequences in (W)STRING

mx990, 2025-05-21:
The `$L` or *Line Feed* escape sequence in (W)STRING is currently assigned the hexadecimal value `0x10`, which turns out to be [Data Link Escape](https://en.wikipedia.org/wiki/Data_Link_Escape). The correct value would be decimal `10`, hexadecimal `0x0a`, or just `'\n'`.

Furthermore, both (W)STRING and (W)CHAR currently translate the character value `'\n'` back to `$N`. The IEC 61131-3 standard 3rd edition defines `$N` or *Newline* as an "implementation-independent means of defining the end of a line of data". That would seem to imply that `$N` is merely an alias and `'\n'` should instead canonically translate to `$L`.

Finally, I would suggest to keep just `'\n'` (LF) as our implementation-defined value for `$N` on all platforms. This also ensures that `$N` can be used in `(W)CHAR` literals, since `\r\n` (CRLF) would simply not fit.

ernstblechaPT, 2025-05-22:
From what i could find online there seems to be some disagreement across PLC Vendors (e.g. Siemens and Rockwell treat `$N` as `CR+LF`, Beckhoff treats `$N` as `LF`) - Does anyone know how Siemens/Rockwell get around the `(W)CHAR` problem?

 - https://literature.rockwellautomation.com/idc/groups/literature/documents/rm/1756-rm014_-en-p.pdf
 - https://support.industry.siemens.com/cs/mdm/109054417?c=63689840011&lc=en-AZ
 - https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_plc_intro/2529327243.html&id=

azoitl, 2025-05-22:
> From what i could find online there seems to be some disagreement across PLC Vendors (e.g. Siemens and Rockwell treat `$N` as `CR+LF`, Beckhoff treats `$N` as `LF`) - Does anyone know how [...]

Is it of our concern how other PLCs are internally treating $N? Because for us the IEC 61131-3 code is important and there it is in all cases $N.

azoitl, 2025-05-22:
> As `$N` is clearly vendor-specific, I would also suggest to either use `$L` or `$R$L` where the actual value or interoperability is important.

Interoperability would anyhow be handled somewhere else: in the communication layers talking with these systems.

MartinMelikMerkumians, 2025-05-22:
> The only reason why I think it might be relevant how other PLCs treat `$N` would be to avoid user confusion. As there are examples for both interpretations there is no "better" solution from that angle.
> 
> Also: as the old situation (`$L` mapping to `0x10`) has never led to a problem I would guess that this is anyways a rather little used feature of Structured Text.

Mapping $L to 0x10 instead to 0x0a violates IEC 61131-3, so I do not think we should keep it that way,

ernstblechaPT, 2025-05-23:
Yes, I fully agree: the mapping of $L should be fixed, the user confusion argument is already a lost cause.

closed 2025-06-24 by mx990