eclipse-4diac/4diac-forte #521: New IO module implementation

pedro-ricardo, 2025-06-27:
Hello there, I have a question about adding a new IO module.

We are starting to setup IOs of a Stahl PLC inside Forte and in this first moment we will implement it with the PARAMS approach due to the apparent simplicity of it.
So far I've understood that we need an implementation of a `CProcessInterfaceBase`  class and for `IX` and `QX` function blocks the methods `initialise`, `deinitialise`, `writePin` and `readPin` should be present.
Of course `CMakeLists` has some settings as well.

I'm willing to know if there is something else I should worry about and witch one of the already existing IO modules in Forte is easier for basing our implementation on?

Thanks in advance

azoitl, 2025-06-27:
If you have a modular IO setup like many PLCs I would strongly suggest to use the modular IO concept. It is code wise cleaner and better extensible. A very good reference is the Wago implementation or the revolution pi. As you can see it is not much code.

pedro-ricardo, 2025-06-28:
Thanks for the advice @azoitl, I'll take a look for sure.

At first sight the IO_Config approach looks far more complex than the other but it indeed fits better with this Stahl PLC we are using.

Do you have any other source documentation that shows how this modular concept works and what are the must have classes and methods to implement ?

azoitl, 2025-06-28:
Especially the automatic event generation process that is backed into the modular IO infrastructure is very handy for reducing 1499 application effort. 

Yes there is more text. Mainly the original bachelor's thesis from which the code originated. I will send you this privately. I by now have the permission from the student to use the bachelor's thesis as basis for a page in the Eclipse 4diac documentation. However haven't found time yet. Maybe someone could support me here.

azoitl, 2025-06-29:
I will move this discussion to out discussion forum where I think it is better placed.